ckb-next/ckb-next #1077: K70 Pro MINI Wireless

Stevenzocker, 2024-07-17:
### What kind of device is it?

Keyboard

### What is the name of the device?

K70 Pro MINI Wireless

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-9189010-na/k70-pro-mini-wireless-60-mechanical-cherry-mx-red-switch-keyboard-with-rgb-backlighting-black-ch-9189010-na

### Generated Report

Drag your generated report here.

gravityfargo, 2024-08-05:
Hi,
Please [generate](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#generate-a-report) and submit a report by attaching it to this issue or opening another.

closed 2024-08-05 by gravityfargo

constambeys, 2024-10-05:
[ckb-next-report-1728136861.gz](https://github.com/user-attachments/files/17266465/ckb-next-report-1728136861.gz)