ckb-next/ckb-next #1078: Darkstar mmo mouse support

julusp, 2024-07-18:
### What kind of device is it?

Mouse

### What is the name of the device?

DARKSTAR WIRELESS RGB MMO Gaming Mouse

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/gaming-mouse/ch-931a011-na/darkstar-wireless-rgb-mmo-gaming-mouse-ch-931a011-na

### Additional Details

2 connection events in report.
1st) stock (freshly unboxed mouse)
2nd) Running the latest FW as of 18th July 2024, updated via Icue on Mac and configured the mouse on it.

### Generated Report

Report generated while plugged over wire.
[ckb-next-report-1721296947.gz](https://github.com/user-attachments/files/16281388/ckb-next-report-1721296947.gz)

pkly, 2024-09-07:
I would also absolutely love this, as I have the same mouse.

spherous, 2024-09-21:
Just got this mouse, would love if I could customize the rgb and bindings :)

patvdleer, 2024-11-28:
My version, out of the box, untouched by any Corsair software, let alone windows.

[ckb-next-report-1732789532.gz](https://github.com/user-attachments/files/17946593/ckb-next-report-1732789532.gz)

patvdleer, 2024-11-29:
# scan codes

These are dumped when connected via usb/wire

## top
### left click
```
ffff8e8c401f0a80 857616342 C Ii:1:012:2 0:1 15 = 01000000 00000000 00000300 000000
ffff8e8c401f0a80 857616377 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 857617283 C Ii:1:012:2 0:1 15 = 01000000 00000000 00000363 000000
ffff8e8c401f0a80 857617299 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 857805325 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 857805363 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 859169380 C Ii:1:012:2 0:1 15 = 01000000 00000000 00000300 000000
ffff8e8c401f0a80 859169424 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 859170350 C Ii:1:012:2 0:1 15 = 01000000 00000000 0000030f 000000
ffff8e8c401f0a80 859170358 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 860313431 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 860313499 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 860748476 C Ii:1:012:2 0:1 15 = 01000000 00000000 00000300 000000
ffff8e8c401f0a80 860748530 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 860749436 C Ii:1:012:2 0:1 15 = 01000000 00000000 0000036d 000000
ffff8e8c401f0a80 860749464 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 861439472 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 861439525 S Ii:1:012:2 -115:1 15 <
```

## right click
```
ffff8e8c401f0a80 875384566 C Ii:1:012:2 0:1 15 = 02000000 00000000 00000300 000000
ffff8e8c401f0a80 875384603 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 875513995 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 875514047 S Ii:1:012:2 -115:1 15 <
```

## scroll click
```
ffff8e8c47d29680 1240165966 C Ii:1:013:2 0:1 15 = 04000000 00000000 00000300 000000
ffff8e8c47d29680 1240166002 S Ii:1:013:2 -115:1 15 <
ffff8e8c47d29680 1240379943 C Ii:1:013:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c47d29680 1240380006 S Ii:1:013:2 -115:1 15 <
```

## left front
```
ffff8e8c401f0a80 1106609896 C Ii:1:012:2 0:1 15 = 00000000 ffff0000 00000300 000000
ffff8e8c401f0a80 1106609945 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 1106684770 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1106684792 S Ii:1:012:2 -115:1 15 <
```

## left rear
```
ffff8e8c401f0a80 1108647993 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1108648026 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 1108762986 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1108763014 S Ii:1:012:2 -115:1 15 <
```

## top middle (under the scroll) front
```
ffff8e8c401f0a80 1174064054 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1174064079 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 1174164392 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1174164418 S Ii:1:012:2 -115:1 15 <
```

## top middle (under the scroll) rear
```
ffff8e8c401f0a80 1199687458 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
[... 49 lines cut ...]
ffff8e8c401f0a80 1054860609 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1054860638 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f1800 1054861616 C Ii:1:012:1 0:1 15 = 01000000 00000100 00000000 000000
ffff8e8c401f1800 1054861649 S Ii:1:012:1 -115:1 15 <
ffff8e8c401f1800 1054962758 C Ii:1:012:1 0:1 15 = 01000000 00000000 00000000 000000
ffff8e8c401f1800 1054962829 S Ii:1:012:1 -115:1 15 <
ffff8e8c401f0a80 1054984615 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1054984630 S Ii:1:012:2 -115:1 15 <
```

### 4 rear bottom
```
ffff8e8c401f0a80 1017793589 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1017793629 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f1800 1017795271 C Ii:1:012:1 0:1 15 = 01000000 00000200 00000000 000000
ffff8e8c401f1800 1017795367 S Ii:1:012:1 -115:1 15 <
ffff8e8c401f0a80 1017883606 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1017883647 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f1800 1017896279 C Ii:1:012:1 0:1 15 = 01000000 00000000 00000000 000000
ffff8e8c401f1800 1017896351 S Ii:1:012:1 -115:1 15 <
ffff8e8c401f0a80 1017910253 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1017910278 S Ii:1:012:2 -115:1 15 <
ffff8e8c401f0a80 1017994260 C Ii:1:012:2 0:1 15 = 00000000 00000000 00000300 000000
ffff8e8c401f0a80 1017994285 S Ii:1:012:2 -115:1 15 <
```

Miastheyna, 2025-11-03:
i am also very interested in this please