ckb-next/ckb-next #1081: Corsair CORSAIR K70 CORE RGB Mechanical Gaming Keyboard Not showing up in CKB-Next + typing issue

Altair314, 2024-07-25:
### Describe your situation in detail.

CKB registers my mouse just fine, but I have a K70 keyboard I'd Like to use. In the GUI of CKB it only shows my mouse, but the logs show that the system recognizes it exists. 

Furthermore, with the keyboard often times when I type a character I get it outputted multiple times. So as an example, one press of the "O" key might read as "OO", though this is not an issue exclusive to Linux as this happens on Windows too

### Additional Details

Add any additional details here.
OS: Linux Mint 21.3 x86_64 
 Host: XPS 8910 1.1.12 
Kernel: 5.15.0-051500-lowlatency 
Uptime: 25 mins 
Packages: 2519 (dpkg), 18 (flatpak) 
Shell: bash 5.1.16 
Resolution: 1920x1080, 1360x768 
DE: Cinnamon 6.0.4 
WM: Mutter (Muffin) 
WM Theme: Mint-Y-Dark-Aqua (Mint-Y) 
Theme: Mint-Y-Dark-Aqua [GTK2/3] 
Icons: Mint-Y-Sand [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i7-6700 (8) @ 4.000GHz 
GPU: Intel HD Graphics 530 
GPU: NVIDIA GeForce GTX 1060 3GB 
Memory: 3970MiB / 15905MiB 

### Generated Report (if applicable)

Drag your generated report here if it's applicable.
[ckb-next-report-1721889855.gz](https://github.com/user-attachments/files/16371821/ckb-next-report-1721889855.gz)

country1706, 2024-08-02:
About the double or multiple output I had the same issue on Windows and Linux.
On Windows I managed it to work a period of time when I forced iCue to update the firmware even though it was the actual one, even on Linux.
Then I found this and changed my keyboard, until now it works fine.
Sorry, it's in german, maybe you can translate it by google.
https://help.corsair.com/hc/de/articles/25038906108429-K70-Core-Doppelte-Tasteneingabe

gravityfargo, 2024-08-05:
Here's the [English version. ](https://help.corsair.com/hc/en-us/articles/25038906108429-K70-Core-Double-Typing) This seems to be a known issue unrelated to ckb-next.

closed 2024-08-05 by gravityfargo