ckb-next/ckb-next #1084: Corsair M65 RGB ULTRA Not Detected

WalkingGlitch, 2024-08-04:
### Affected Devices

Corsair M65 RGBA ULTRA

### Linux Distribution

MX Linux 23.3

### Installation Method

Compiled from Git Main

### ckb-next version

ckb-next 0.6.0-48-gecde09e

### What did you expect?

Expected behavior: Mouse should be detected and display in the GUI.

Actual behavior: Displays "No devices connected"

### Did you try anything that may have helped?

Tried rebooting and tried running as root. Running as root with sudo does not work, the application doesn't let you. I also had issues generating a report. I had to do so manually from a new terminal window, using sudo. I will file another bug ticket for this.

### Additional Details

Generating a report had to be done manually from a sudo shell. The report correctly lists the mouse. The mouse is usable in the OS, and it is correctly identified via lsusb. OS is running the most up to date packages at the time of this ticket.

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/16488867/ckb-next-dev-detect-report.gz)

WalkingGlitch, 2024-08-04:
I've found this was my fault. I got tricked by Corsair's naming convention. Looking into the issue further, it seems mine is not the M65 RGB listed on the supported list, but is instead the M65 ULTRA. It may be helpful to add more specificity to the list to prevent this kind of misunderstanding. I'm going to close the ticket as such.

closed 2024-08-04 by WalkingGlitch

Ravenslofty, 2024-08-04:
We use as much specificity as Corsair do. It's not much easier on our end when, for example, Corsair made three mice all called the Sabre.

WalkingGlitch, 2024-08-09:
I get it. It was the semantics of how they label it on the packaging that threw me. I thought it was the RGB referenced on the device table, and not the Ultra, in thinking that the Ultra would be the "M65 ULTRA xyz" instead of the "M65 RGB ULTRA." I only realized my error when I forked the repo to do my own investigating and matched the device ID strings. Oops.