ckb-next/ckb-next #1085: K100 Erratic Wheel Behavior

kimjongbing, 2024-08-06:
### Affected Devices

Corsair K100 RGB Mechanical Gaming Keyboard

### Linux Distribution

Arch Linux

### Installation Method

AUR

### ckb-next version

ckb-next 0.6.0-48-gecde09ef

### What did you expect?

The wheel key should do nothing by default or be bindable to something else. Instead, if you touch the wheel key it will start spamming random characters or try to use the super/alt/ctrl keys. It will go sicko mode.

### Did you try anything that may have helped?

I tried to bind it to something or do anything, but there is no changing its behavior.

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1722932085.gz](https://github.com/user-attachments/files/16507434/ckb-next-report-1722932085.gz)

gravityfargo, 2024-09-19:
This keyboard is not currently supported. See #651

closed 2024-09-19 by gravityfargo

kimjongbing, 2024-09-19:
> This keyboard is not currently supported. See #651

Hi, this does not make sense though. It is working on Arch but not on NixOS. Is there a possibility there is a discrepancy between the two packages from the AUR and Nixpkgs?