ckb-next/ckb-next #1086: Corsair K100 new Product ID 1b7c

merlinlcb, 2024-08-18:
### What kind of device is it?

Keyboard

### What is the name of the device?

CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard

### Additional Details

it appears theres a new product ID for the k100 optical 0x1b7c, i was able to get it ISH working by adding it to src/daemon/usb.h however it absolutely freaks out and spams whatever key you press until unplugged

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/16646772/ckb-next-dev-detect-report.gz)

i have BOTH a working one and the new model plugged in, a 1b7c and a 1bc5

gravityfargo, 2024-09-19:
1b7c has been requested with #651 but 1bc5 is not in the wiki. I have added it.

Arghadharg, 2025-12-19:
anyone got an update on this?