ckb-next/ckb-next #1088: K100 not detected

MakersMarked, 2024-08-18:
### Affected Devices

CORSAIR K100 RGB Mechanical Gaming Keyboard

### Linux Distribution

Fedora

### Installation Method

dnf isntall

### ckb-next version

ckb-next 0.6.0

### What did you expect?

Keyboard should be detected, as its a corsair keyboard. 

It has previously worked, though I don't remember if I was on Fedora or Ubuntu when I last used ckb-next(same hardware)

### Did you try anything that may have helped?

installed then rebooted computer

unplugged and plugged back in

unplugged both mouse and keyboard and tried just the keyboard

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/16650812/ckb-next-dev-detect-report.gz)

coldkick2, 2024-08-28:
I am also not having my K100 detected.

Linux Distribution
Arch

Installation Method
yay -S ckb-next

ckb-next version
ckb-next 0.6.0

RyanEskridge, 2024-09-09:
Strange. It looks like its having trouble connecting to the mouse. Can someone comment if its normal for the ckb daemon to connect and disconnect the a peripheral so many times?

Can you try without the mouse plugged in?

kimjongbing, 2024-09-10:
> It has previously worked, though I don't remember if I was on Fedora or Ubuntu when I last used ckb-next(same hardware)
I have similarly inconsistent behavior. On CachyOS/Arch it is detected but on NixOS it is not
ckb-next 0.6.0

ckb-next-dev-detect - Tue Sep 10 10:48:23 AEST 2024

```
Kernel: Linux nixos 6.10.8-cachyos #1-NixOS SMP PREEMPT_DYNAMIC Wed Sep  4 11:30:16 UTC 2024 x86_64 GNU/Linux

OS Version:

USB Devices:
/nix/store/ggn20rbx8lrhjwp9rj5la3irh93sng7s-ckb-next-0.6.0/bin/ckb-next-dev-detect: line 60: lsusb: command not found

Dmesg on current boot:
Sep 10 10:41:15.486579 nixos kernel: usb 3-1: New USB device found, idVendor=1b1c, idProduct=1b7c, bcdDevice= 0.20
Sep 10 10:41:15.486779 nixos kernel: usb 3-1: Product: CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard
Sep 10 10:41:15.486876 nixos kernel: usb 3-1: Manufacturer: Corsair
Sep 10 10:41:16.130576 nixos kernel: input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard as /devices/pci0000:00/0000:00:08.1/0000:29:00.3/usb3/3-1/3-1:1.0/0003:1B1C:1B7C.0001/input/input2
Sep 10 10:41:16.184578 nixos kernel: input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard Mouse as /devices/pci0000:00/0000:00:08.1/0000:29:00.3/usb3/3-1/3-1:1.0/0003:1B1C:1B7C.0001/input/input3
Sep 10 10:41:16.184595 nixos kernel: hid-generic 0003:1B1C:1B7C.0001: input,hidraw0: USB HID v1.00 Keyboard [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input0
Sep 10 10:41:16.184755 nixos kernel: hid-generic 0003:1B1C:1B7C.0002: hiddev96,hidraw1: USB HID v1.00 Device [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input1
Sep 10 10:41:16.185029 nixos kernel: hid-generic 0003:1B1C:1B7C.0003: hiddev97,hidraw2: USB HID v1.00 Device [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input2

Dmesg on last boot:
Sep 10 10:18:42.595055 nixos kernel: usb 3-1: New USB device found, idVendor=1b1c, idProduct=1b7c, bcdDevice= 0.20
Sep 10 10:18:42.595252 nixos kernel: usb 3-1: Product: CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard
Sep 10 10:18:42.595350 nixos kernel: usb 3-1: Manufacturer: Corsair
Sep 10 10:18:43.351469 nixos kernel: input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard as /devices/pci0000:00/0000:00:08.1/0000:29:00.3/usb3/3-1/3-1:1.0/0003:1B1C:1B7C.0001/input/input2
Sep 10 10:18:43.407470 nixos kernel: input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard Mouse as /devices/pci0000:00/0000:00:08.1/0000:29:00.3/usb3/3-1/3-1:1.0/0003:1B1C:1B7C.0001/input/input3
Sep 10 10:18:43.407491 nixos kernel: hid-generic 0003:1B1C:1B7C.0001: input,hidraw0: USB HID v1.00 Keyboard [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input0
Sep 10 10:18:43.407672 nixos kernel: hid-generic 0003:1B1C:1B7C.0002: hiddev96,hidraw1: USB HID v1.00 Device [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input1
Sep 10 10:18:43.407818 nixos kernel: hid-generic 0003:1B1C:1B7C.0003: hiddev97,hidraw2: USB HID v1.00 Device [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input2
Sep 10 10:33:53.715477 nixos kernel: usb 3-1: New USB device found, idVendor=1b1c, idProduct=1b7c, bcdDevice= 0.20
Sep 10 10:33:53.715812 nixos kernel: usb 3-1: Product: CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard
Sep 10 10:33:53.715921 nixos kernel: usb 3-1: Manufacturer: Corsair
Sep 10 10:33:53.863473 nixos kernel: input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard as /devices/pci0000:00/0000:00:08.1/0000:29:00.3/usb3/3-1/3-1:1.0/0003:1B1C:1B7C.000B/input/input29
Sep 10 10:33:53.917469 nixos kernel: input: Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard Mouse as /devices/pci0000:00/0000:00:08.1/0000:29:00.3/usb3/3-1/3-1:1.0/0003:1B1C:1B7C.000B/input/input30
Sep 10 10:33:53.931468 nixos kernel: hid-generic 0003:1B1C:1B7C.000B: input,hidraw0: USB HID v1.00 Keyboard [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input0
Sep 10 10:33:53.931668 nixos kernel: hid-generic 0003:1B1C:1B7C.000C: hiddev96,hidraw1: USB HID v1.00 Device [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input1
Sep 10 10:33:53.933469 nixos kernel: hid-generic 0003:1B1C:1B7C.000D: hiddev97,hidraw2: USB HID v1.00 Device [Corsair CORSAIR K100 RGB Optical-Mechanical Gaming Keyboard] on usb-0000:29:00.3-1/input2

Daemon on current boot:
-- No entries --

Daemon on last boot:
-- No entries --

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/connected
/dev/input/ckb0/version
0.6.0
/dev/input/ckb0/pid
1123

Kernel command line:
initrd=\EFI\nixos\bzvrqsrhddg1rlcqmg1yhdn895m88dc7-initrd-linux-6.10.8-initrd.efi init=/nix/store/39zjz284dajqpwkjrbdprb66q4rf2gaj-nixos-system-nixos-24.11.20240906.574d1ea/init loglevel=4 nvidia-drm.modeset=1 nvidia-drm.fbdev=1

GUI ldd:
	linux-vdso.so.1 (0x00007a0c6243b000)
	libc.so.6 => /nix/store/wlffq5p6mxxgfap10sav3ij936jzqm59-glibc-2.39-52/lib/libc.so.6 (0x00007a0c6223e000)
	/nix/store/wlffq5p6mxxgfap10sav3ij936jzqm59-glibc-2.39-52/lib/ld-linux-x86-64.so.2 => /nix/store/wlffq5p6mxxgfap10sav3ij936jzqm59-glibc-2.39-52/lib64/ld-linux-x86-64.so.2 (0x00007a0c6243d000)

Environment:
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=wayland
QT_BEARER_POLL_TIMEOUT=-1
XDG_CURRENT_DESKTOP=GNOME
WAYLAND_DISPLAY=wayland-0
[... cut ...]
NIX_XDG_DESKTOP_PORTAL_DIR=/run/current-system/sw/share/xdg-desktop-portal/portals
QT_PLUGIN_PATH=/nix/store/07z7ibaapmfxa37zkvg8vvm1arm80qp1-qttools-5.15.14-bin/lib/qt-5.15.14/plugins:/nix/store/qk8x80js28pd3c62s8nzafag26g63wqm-qtwayland-5.15.14-bin/lib/qt-5.15.14/plugins:/nix/store/rq63n5jhhcjpjh32vixddrfi9kpr4b45-qtdeclarative-5.15.14-bin/lib/qt-5.15.14/plugins:/nix/store/bqy8jzvpg6djvvfwmhcdx34k9m0ilxck-qtsvg-5.15.14-bin/lib/qt-5.15.14/plugins:/nix/store/bc6f63sli7xqwb5467ka5a39m2n4xfgs-qtbase-5.15.14-bin/lib/qt-5.15.14/plugins
```

gravityfargo, 2024-09-19:
So this keyboard was added on  Jun 23, 2023 with ed70884da8c66c9ea9fb08c30f37fbcc67a250cb but there is no issue asociated with it. It's listed in the wiki but it's not requested or listed as supported. I'm going to reclasify this as a new device request pending a dev's input.

phxMurray, 2024-12-15:
anyone happen to get 1b7d working?

gravityfargo, 2025-05-13:
added in ed70884 with release 0.6.1/0.6.2

closed 2025-05-13 by gravityfargo