ckb-next/ckb-next #1089: Add support for HX1000i Power Supply

shoulders, 2024-08-21:
### What kind of device is it?

Power Supplies

### What is the name of the device?

HX1000

### Link to Corsair's Product Page

https://www.corsair.com/uk/en/p/psu/cp-9020259-uk/hx1000i-fully-modular-ultra-low-noise-platinum-atx-1000-watt-pc-power-supply-cp-9020259-uk

### Additional Details

I would like to use my HX1000i with your driver in TrueNAS and also NetData for monitoring the power supply.

https://forums.truenas.com/t/monitor-smart-pc-power-supplies-corsair-icue-and-others/11465/2

### Generated Report

n/a

SethBarberee, 2024-08-28:
Assuming your kernel is up-to-date, there is `hwmon`/kernel support in the `corsair-psu` module:
https://docs.kernel.org/hwmon/corsair-psu.html

In addition, `liquidctl` has support on top of the kernel module for monitoring:
https://github.com/liquidctl/liquidctl/blob/main/docs/corsair-hxi-rmi-psu-guide.md

gravityfargo, 2024-09-19:
Powersupply is outside of the scope of this project, and it looks like the previous commenter suggested a good resource.

closed 2024-09-19 by gravityfargo