ckb-next/ckb-next #1092: Cannot find devices

kelpc100, 2024-08-23:
### Describe your situation in detail.

I am having trouble finding my devices. I am using the K70 RGB Pro Keyboard and Scrimtar RGB Elite Wireless Mouse, but the devices will not show up. I have tried unplugging/plugging back in, restarting the program, restarting my computer, reinstalling ckb-next, and re installing linux but the devices will not show up in ckb-next. I am using Linux Mint 22 cinnamon, and tried downloading via the command line (v 0.6.0) and software manager (v 0.6.0). I have used lusb to check if my devices are showing up, and they appear as normally, yet no luck with the software. Any help would be appreciated.

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

DreiPixel, 2024-08-24:
they not supported yet.
https://github.com/ckb-next/ckb-next/pull/1030
https://github.com/ckb-next/ckb-next/pull/997

kelpc100, 2024-08-28:
Ok, thank you.

closed 2024-08-28 by kelpc100