ckb-next/ckb-next #1094: Scimitar RGB Elite 0x1be3 not work

AntonioOri, 2024-09-10:
### Describe your situation in detail.

Hello, 

Adquired a new Scimitar RGB Elite, It didn't work, maybe new idProduct?

Arch updated
Test witch ckb-next stable, git, and with --enabled-experimental
Added usbcore.quirks=1b1c:1be3:gn to kernel parameters, but no luck.

Thanks

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.
[ckb-next-report-1725968437.gz](https://github.com/user-attachments/files/16945145/ckb-next-report-1725968437.gz)

SunflowerFuchs, 2024-09-14:
I'm having the same issue, just got the mouse a few days ago, ckb-next doesn't seem to recognize it. 
Currently running 0.6.0-49-g7708b0c2, also tried experimental as well as adding the kernel param. 
UI errors out when I try to generate a report, so I sadly can't add one.

gravityfargo, 2024-09-19:
This device is not currently supported. I'm reclassifying this as a device request and have added it to the wiki.

ArbitraryAlex, 2024-09-23:
Also having this issue.
ckb-next-report does seem to detect my Scimitar RGB ELITE in the dmesg section, but ckb doesn't list a dev node for it, and there's no menu entry for same.

migmolrod, 2024-11-14:
Interesting. I had an "older" Scimitar RGB Elite (bought in 2021) and it worked flawlessly in ckb-next. I could bind keys, change DPI and change light fx. Now I got a replacement unit and the new one is not detected. Maybe new batches are different from the old ones, firmware wise.

Everspace, 2024-11-28:
I'm wondering if this is also solved by https://github.com/ckb-next/ckb-next/pull/997

jacobluke89, 2024-12-18:
I'm having issues with Scimitar RGB Elite. I am running 24.04 LTS, could this be the route of my issues? Should I switch to 22.04? I'm interested in using the macro buttons in an IDE. Thanks

gammexane, 2025-03-11:
> [ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/18829788/ckb-next-dev-detect-report.gz)
> 
> Is there anything needed information wise about this model?
> 
> My old Scimitar RGB Pro just died on me, first the mouse wheel stopped clicking which drove me nuts so I opened and cleaned it just to find that the sensor is now completely dead :(
> 
> Raced to the store to buy a new one, just to find this model isn't yet supported!

Exactly the same here, I just bought the third Scimitar RGB just because it was supported great on Linux by CKB-NEXT...  but this one it is not... 

Any update on this?

jacobluke89, 2025-03-11:
I've found this model works on 24 ubuntu when set up using corsair's
microsoft's software. Unfortunately I am yet to get it working with
CKB-NEXT.

gammexane, 2025-03-11:
I found an alternative open source project that let me configure the mouse: https://github.com/jurkovic-nikola/OpenLinkHub 
Maybe you can check it and make the RGB ELITE work on CKB-NEXT.
 
At least I could map the keys and change the rainbow color to static :D

Yueh72, 2025-03-29:
Same thing here, I just buy a scimitar elite as it's in supported device to discover wich is not in fact...

nilbest, 2025-04-03:
I have the Same Problem.
It does not get detected in ckb-next.
At least the mouse is recognized as a mouse on Linux.
But setting up the key would be nice.

scott-dft, 2025-07-19:
To confirm detail as @BetrayedMussel stated.
A new Wired CORSAIR SCIMITAR RGB ELITE not detected;
With report giving the product as:
```
idVendor           0x1b1c Corsair
idProduct          0x1be3 CORSAIR SCIMITAR RGB ELITE Gaming Mouse
```

Falson127, 2025-07-20:
I just got one of these. I used to get by with a windows vm to configure the hardware profiles with iCUE and then use the mouse on linux like normal, but for some reason this particular device does not properly save its hardware profiles. They only work when iCUE is running and the mouse is set to "onboard memory mode" which is of no use.

Can anyone point me towards the tools I would need to use to try and add support for this device to ckb-next? Corsair is basically the only option for an MMO mouse on Linux, and they no longer seem to sell the old supported versions of the Scimitar line.

sfjuocekr, 2025-09-27:
> I just got one of these. I used to get by with a windows vm to configure the hardware profiles with iCUE and then use the mouse on linux like normal, but for some reason this particular device does [...]

Actually, you can save to the onboard memory and it will be active outside of iCUE! You can tell by the touch pad being colored red.

BetrayedMussel, 2026-03-23:
The `idProduct` 0x1b8b corresponds to the older Scimitar RGB Elite V1, which the current code already detects.  

Corsair’s newer Scimitar RGB Elite V2 uses `idProduct` 0x1be3, which is not detected in the latest code without #1275.  

I have both devices: with #1275 applied, both the V1 (0x1b8b) and V2 (0x1be3) are detected as separate configurable items. Without #1275, only the older V1 (0x1b8b) is recognized.

tatokis, 2026-03-23:
Sorry, I'm genuinely not trying to be annoying, but are you 100% sure it doesn't work with the latest code from master and without the PR?

The newer `0x1be3` is already there, as you can see, as `P_SCIMITAR_ELITE_BRAGI`. And if you look at all the [changes](https://github.com/ckb-next/ckb-next/pull/1275/changes) in that PR, `P_SCIMITAR_ELITE_BRAGI` is already there in all those modified areas/defines except `USES_FILE_HWSAVE` which isn't needed for bragi devices as hwload is only implemented for NXP. Doing Ctrl-F `P_SCIMITAR_ELITE_BRAGI` in the changes page shows that it's there.

I just don't see how that PR causes any change whatsoever to make it work.

RjFnC, 2026-08-16:
I have a Corsair Scimitar Elite RGB that was bought Feb 2026 and it's not showing up in CKB. Here is the Debug Report.
[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/31112993/ckb-next-dev-detect-report.gz)

Thank you.

vochilan, 2026-09-23:
Any chance support for 2b22 will be buildable anytime soon? I was not aware my new $80 mouse would be mostly function-less upon arrival :(