ckb-next/ckb-next #1097: GUI crashes: "The Wayland connection experienced a fatal error: Protocol error"

atomflunder, 2024-09-24:
### Affected Devices

K70

### Linux Distribution

Fedora

### Installation Method

Gnome Software / RPM

### ckb-next version

0.6.0

### What did you expect?

Opening the GUI via a shortcut or with the terminal using the `ckb-next` command crashes the program immediately, with the following error message:

```bash
name@pc:~$ ckb-next
QSocketNotifier: Can only be used with threads started with QThread
ckb-next 0.6.0
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/name/.config/ckb-next/ckb-next.conf"
QFont::setPixelSize: Pixel size <= 0 (0)
The Wayland connection experienced a fatal error: Protocol error
The Wayland connection experienced a fatal error: Protocol error
```

When I open the program in the background with `ckb-next --background`, everything works fine, just the GUI is crashing for me.

### Did you try anything that may have helped?

On a previous Fedora installation, with the same hardware and the same ckb-next version the issue did not come up. Was also using Wayland back then, I do not think anything major on my changed.

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/17117953/ckb-next-dev-detect-report.gz)

tatokis, 2024-09-26:
Judging by the message, I suspect the issue is with Qt or the compositor. You can try

1. Running under X11 with `QT_QPA_PLATFORM=xcb ckb-next`
2. Running under a different wayland compositor
3. Building ckb-next manually from git
4. Building ckb-next manually from git against Qt6

Don't think there's much I can do here.

tatokis, 2024-09-26:
I found the following two issues that might be related; if you can reproduce them then it's possible the root cause is the same.

https://bugs.kde.org/show_bug.cgi?id=488090
https://code.videolan.org/videolan/vlc/-/issues/28679

atomflunder, 2024-09-26:
Hi @tatokis, thanks for the quick response.

I haven't found that kleopatra thread while googling the problem, but I can confirm it also happens on my machine - the GUI won't even launch at all. I couldn't find anything similar to the issue, and the crash was only affecting ckb-next, so I opened the issue here, apologies if it has nothing to do with the app itself. If you think this is completely out of scope, feel free to close the issue.

Running `ckb-next` under X11 with your provided command fixes the issue.  
Come to think of it, the only meaningful difference between my old setup (where ckb-next was working fine), and my current setup is that on my old setup I used to run X11 and later upgraded to Wayland, while on my new setup I ran Wayland straight away. Maybe that has something to do with it?

tatokis, 2024-09-27:
> apologies if it has nothing to do with the app itself

No worries! At the very least I am now aware that this is an issue and can guide people on how to work around it.

Only reason I suspect it has nothing to do with ckb-next is because we don't do anything with wayland directly. Unfortunately this also means I likely can't fix it.

> Come to think of it, the only meaningful difference between my old setup (where ckb-next was working fine), and my current setup is that on my old setup I used to run X11 and later upgraded to Wayland, while on my new setup I ran Wayland straight away. Maybe that has something to do with it?

I think it's unlikely.

I have very limited experience, but it might be worth trying to run `WAYLAND_DEBUG=1 ckb-next` and attaching the logs. Might shed some light on why this happens.

atomflunder, 2024-09-27:
> I have very limited experience, but it might be worth trying to run WAYLAND_DEBUG=1 ckb-next and attaching the logs. Might shed some light on why this happens.

@tatokis  I ran the command and I think this provides a great clue, Here is the [whole log file](https://github.com/user-attachments/files/17166017/logs.txt) but I think the last line is what matters here:

```bash
[ 117985.392] {Display Queue} wl_display#1.error(wp_linux_drm_syncobj_surface_v1#46, 2, "Explicit Sync only supported on dmabuf buffers")
The Wayland connection experienced a fatal error: Protocol error
```

When I search for that exact phrase the only result is the [actual source code for Mutter](https://git.saltyming.net/sungmg/mutter-performance-source/commit/e8b890ab53a67da2dcb9a9993f143a58c403daa1.patch), but I also found some similar issues [here (Nvidia Wayland Repo)](https://github.com/NVIDIA/egl-wayland/issues/110) and [here (Firefox Bug Tracker)](https://bugzilla.mozilla.org/show_bug.cgi?id=1908825)

[This issue](https://github.com/Soundux/Soundux/issues/709) is also related, and I can confirm that running `__NV_DISABLE_EXPLICIT_SYNC=1 ckb-next` also works without crashing.

I think this looks like an issue with Qt. I haven't yet tested your suggestion about building `ckb-next`with Qt6, how would I be able to do that? Maybe that could fix it.

atomflunder, 2024-10-02:
Small, probably meaningless, update:  
I installed the COSMIC alpha today, and the crashes do not happen there, also running Wayland. I guess it is specific to Mutter / Gnome. Haven't tried KDE yet, I would guess it also functions fine there since the error message is coming from Mutter itself.

tatokis, 2024-10-12:
Thank you for figuring out the workaround. It really looks like there's not much we can do.

It's probably a bad interaction between the nvidia driver and mutter.

I'll go ahead and close this, but I'll pin it so that others can find it easily. Thanks :)

closed 2024-10-12 by tatokis

masterdokuro, 2024-10-29:
Thanks for pinning this issue. I found it really quickly when I ran into it myself yesterday.

My environment is fedora 40 with 560.35.03 nvidia drivers + wayland + K100 keyboard.  As another workaround you can run `XDG_SESSION_TYPE=x11 ckb-next` which will not crash.

However, even running in the background if I enable **Turn lights off when idle** in the settings it will cause ckb-next to crash.  To workaround this I disabled that setting and added another profile/mode which is pure black and then I switch to that using the profile button on the keyboard.

masterdokuro, 2025-05-03:
FYI... Interestingly I installed ckb-next (0.6.2-1.fc42) via the fedora repro and this crash problem does not happen.  I can only guess that the package maintainer on fedora has done some voodoo wizardry to fix this problem with wayland.  Anyhow, said I'd post this here in case anyone else finds this useful.

tatokis, 2025-05-26:
I suspect it's either Qt6, or there was an nvidia update that fixed it. Either way, I am slowly moving all packages/installation methods towards Qt6.