ckb-next/ckb-next #1106: ckb-next crashes on Wayland

gammexane, 2024-10-29:
### Affected Devices

all

### Linux Distribution

Fedora 41

### Installation Method

dnf5

### ckb-next version

ckb-next 0.6.0

### What did you expect?

I have just installed Fedora 41 and ckb-next crashes when I try to use it:

[$] ckb-next
QSocketNotifier: Can only be used with threads started with QThread
ckb-next 0.6.0
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/<user>/.config/ckb-next/ckb-next.conf"
QFont::setPixelSize: Pixel size <= 0 (0)
QFont::setPixelSize: Pixel size <= 0 (0)
QFont::setPixelSize: Pixel size <= 0 (0)
The Wayland connection experienced a fatal error: Protocol error

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
I can't generate it. The app crashes.

gammexane, 2024-10-29:
I found another report about this: 1097

Basically, the workaround is to run ckb-next as: `__NV_DISABLE_EXPLICIT_SYNC=1 ckb-next`

Kangaroux, 2024-11-26:
#1097

arnaud-sk, 2024-11-26:
Hello, I am experiencing the same problem and I do not understand where to put this command `__NV_DISABLE_EXPLICIT_SYNC=1 ckb-next`. 
If I do it in a terminal, there is no problem, but if I close the terminal, it also closes ckb-next. Also, how to get ckb-next to use this how when starting the session?

gravityfargo, 2024-12-11:
@arnaud-sk If you're starting via your  DE, be it KDE or gnome, you can modify the .desktop file. I've added a [wiki](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#crash-on-wayland-with-nvidia-gpus) entry for it.  For instance with KDE, using "Autostart" from the system setting uses the desktop file. 
[image]

Matheos96, 2024-12-31:
Experienced the same issue on Fedora WS 41 Wayland + GNOME. The workaround worked and I noticed that for GNOME you need to edit two `.desktop` files as the autostart one is separate from the one that you find in the launcher.

TLDR; I had to add `env __NV_DISABLE_EXPLICIT_SYNC=1` to the beginning of the `Exec=` parts in both  
`/usr/share/applications/ckb-next.desktop`  
AND  
`~/.config/autostart/ckb-next.autostart.desktop`

Matheos96, 2025-01-02:
> AND `~/.config/autostart/ckb-next.autostart.desktop`

Just noticed that somehow this file had gotten rewritten on my machine. This time it was read only and reverted back to its original state. Why? I made it writeable and re-did the workaround but why would it revert? Does not seem like ckb-next has updated on my machine since my last post...

Matheos96, 2025-01-03:
> Just noticed that somehow this file had gotten rewritten on my machine. This time it was read only and reverted back to its original state. Why? 

Answering my own question...

`ckb-next` ensures that the `~/.config/autostart/ckb-next.autostart.desktop` entry is present on each login **if** you have the _start ckb-next at login` setting checked in settings. This will seemingly overwrite it with the default file each time too, thus modifications are lost.

To workaround this, I checked the setting, generating the default file, added write permissions to the file (`chmod +w ~/.config/autostart/ckb-next.autostart.desktop`), modified it according to previous posts and then finally I renamed it (important) to `ckb-next-autostart-fixed.desktop`. Finally I **unchecked** the setting in the GUI. This is important as otherwise it will re-create the original file with the original name, each login. Also, if you don't rename it, the action of unchecking the setting will remove the file, even with your modifications. GNOME will now launch the application using the custom file on login, but the `ckb-next` application does not know anything about it (which is why you should explicitly have the "start at login" setting unchecked now).

arnaud-sk, 2025-01-04:
I follow the [wiki](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#crash-on-wayland-with-nvidia-gpus), but ckb-next continue crash.
I have edit `/usr/share/applications/ckb-next.desktop` and add `env __NV_DISABLE_EXPLICIT_SYNC=1`
```
      │ File: /usr/share/applications/ckb-next.desktop
───────┼─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ # Copyright 2017-2018 ckb-next Development Team <ckb-next@googlegroups.com>
   2   │ # Distributed under the terms of the GNU General Public License v2
   3   │ 
   4   │ [Desktop Entry]
   5   │ Version=1.0
   6   │ Type=Application
   7   │ Name=ckb-next
   8   │ Comment=Corsair keyboard driver user interface
   9   │ Keywords=ckb;corsair;keyboard;rgb;
  10   │ Categories=Settings;System;
  11   │ TryExec=/usr/bin/ckb-next
  12   │ # Exec=/usr/bin/ckb-next
  13   │ Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 /usr/bin/ckb-next
  14   │ Icon=ckb-next
  15   │ StartupNotify=true
```

Matheos96, 2025-01-04:
Sounds like the app you are launching is not the one that you have edited. On gnome I can find the app. Do you possibly have a duplicate in `~/.local/share/applications`? Just guessing

Matheos96, 2025-01-07:
You could try searching your computer for .desktop files which define an application with the name `ckb-next`

`grep -r "Name.*=ckb-next" --include="*.desktop" --exclude-dir=boot,dev,proc,run,snap,sys / 2> /dev/null`

For me this shows my two files, one of which is the one that I find in gnome and the other is used for autostart. If you find any other than the two you know of, maybe that is the issue.

My output:
```
❯ grep -r "Name.*=ckb-next" --include="*.desktop" --exclude-dir=boot,dev,proc,run,snap,sys / 2> /dev/null

/home/matheos/.config/autostart/ckb-next.autostart-fixed.desktop:Name=ckb-next
/usr/share/applications/ckb-next.desktop:Name=ckb-next
```

Jgnosis, 2025-01-10:
same issue with Bazzite (Fedora fork). Universal Blue forum said its the app itself doesn't work with Wayland and here all I've seen is that it is the distro that is causing it.  It launches then crashes the second I click on the GUI whether using KDE or Gnome DE.  I switched back to X11 with a console command but was warned that Fedora will be Wayland only in the near future.  So I'm in the same boat as you.

arnaud-sk, 2025-01-14:
> You could try searching your computer for .desktop files which define an application with the name `ckb-next`
> 
> `grep -r "Name.*=ckb-next" --include="*.desktop" --exclude-dir=boot,dev,proc,run,snap,sys / 2> /dev/null`
> 
> For me this shows my two files, one of which is the one that I find in gnome and the other is used for autostart. If you find any other than the two you know of, maybe that is the issue.
> 
> My output:
> 
> ```
> ❯ grep -r "Name.*=ckb-next" --include="*.desktop" --exclude-dir=boot,dev,proc,run,snap,sys / 2> /dev/null
> 
> /home/matheos/.config/autostart/ckb-next.autostart-fixed.desktop:Name=ckb-next
> /usr/share/applications/ckb-next.desktop:Name=ckb-next
> ```
> 
> Based the command on [this post](https://askubuntu.com/a/1356480/2083854)

I have same output : 

```
> grep -r "Name.*=ckb-next" --include="*.desktop" --exclude-dir=boot,dev,proc,run,snap,sys / 2> /dev/null

/home/arnaud/.config/autostart/ckb-next.autostart.desktop:Name=ckb-next
/usr/share/applications/ckb-next.desktop:Name=ckb-next
```
And in two files, there is `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 /usr/bin/ckb-next`

arnaud-sk, 2025-01-14:
I don't understand. I try change `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 /usr/bin/ckb-next` to `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 ckb-next` in `/usr/share/applications/ckb-next.desktop` and it's work.

It's very strange.

Matheos96, 2025-01-14:
> I don't understand. I try change `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 /usr/bin/ckb-next` to `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 ckb-next` in `/usr/share/applications/ckb-next.desktop` and it's work.
> 
> It's very strange.

:open_mouth:  Great that you got it working but that is indeed quite strange. I am no guru by any means, but I thought thouse would be identical... Unless you somehow have two `ckb-next` installed. What does `whereis ckb-next` return?

This is mine. I guess the second one is man pages and the first one is the binary itself.
```
❯ whereis ckb-next
ckb-next: /usr/bin/ckb-next /usr/share/man/man1/ckb-next.1.gz
```

arnaud-sk, 2025-01-14:
> > I don't understand. I try change `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 /usr/bin/ckb-next` to `Exec=env __NV_DISABLE_EXPLICIT_SYNC=1 ckb-next` in `/usr/share/applications/ckb-next.desktop` and it's work.
> > It's very strange.
> 
> 😮 Great that you got it working but that is indeed quite strange. I am no guru by any means, but I thought those would be identical... Unless you somehow have two `ckb-next` installed. What does `whereis ckb-next` return?
> 
> This is mine. I guess the second one is man pages and the first one is the binary itself.
> 
> ```
> ❯ whereis ckb-next
> ckb-next: /usr/bin/ckb-next /usr/share/man/man1/ckb-next.1.gz
> ```

Same output : 
```
> whereis ckb-next
ckb-next: /usr/bin/ckb-next /usr/share/man/man1/ckb-next.1.gz
```

Matheos96, 2025-01-14:
Sorry mate, we need to consult someone smarter than me if we wanna figure this one out 😅

tatokis, 2025-05-26:
I've had a report that this should be fixed, at least on Fedora. https://github.com/ckb-next/ckb-next/issues/1097#issuecomment-2908619739

If someone can verify, that'd be great. If you're not running fedora, make sure you have the latest nvidia drivers and are using a Qt6 build of ckb-next. You can click the About Qt button in Settings and look at the version number there. It should say Qt 6.x.x.

Some todo items out of this ticket though:
1. ckb-next should not write an autostart file.
2. We should simply install an autostart file in /etc/xdg/autostart and get rid of this manual mechanism. That way, DE editors can manage the autostart item properly.
3. The GUI should parse the autostart file in ~/.config/autostart/ if present, to detect X-GNOME-Autostart-enabled and the KDE method of disabling autostart. Alternatively, the autostart option needs to go completely in ckb-next.