ckb-next/ckb-next #1114: Keystroke double recognition while software is active

milowagnitz, 2024-11-26:
### Affected Devices

K70 RGB Core

### Linux Distribution

Ubuntu 24.04.1

### Installation Method

DEB Ubuntu Repository

### ckb-next version

ckb-next 0.6.0+dfsg

### What did you expect?

Normal recognition of keystrokes. If the software is not running, keystrokes are recognized norrmally. But when it runs, often some keys that are pressed once are being reconized as pressed twice. Writing is much harder with this issue and I have to exit the software to write normally again

### Did you try anything that may have helped?

The keyboard has to be resetted every time Ubuntu is rebooted

### Additional Details

The error often occurs when the keyboard was used in Windows before with activated Corsair iCUE Software. I have to dual boot between Windows 11 and Ubuntu 24.04.1 for software compatibility reasons

### Generated Report

[ckb-next-report-1732633627.gz](https://github.com/user-attachments/files/17921519/ckb-next-report-1732633627.gz)

rskunath, 2024-11-26:
Any possibility that this is the famous key chatter that Corsair mechanical keyboards get after many years of use? Mine started doing that last year on the more used keys. These are MX Speed switches. How old is the keyboard? But maybe this is some different behavior than the chatter?

trashguy, 2024-12-22:
I am experiencing this as well even with this patch #1117