ckb-next/ckb-next #1116: K100 RGB (1b7d) not detected

Kvoo, 2024-12-03:
### Affected Devices

K100 RGB

### Linux Distribution

Fedora KDE

### Installation Method

dnf

### ckb-next version

ckb-next 0.6.0

### What did you expect?

Expected behaviour is that the Keyboard is detected inside of ckb-next and I am able to configure its RGB and settings. As it is, only my Mouse and Mouse Pad are detected. I also have corsair fans, LED strips, Dominator RAM, an AIO and a headset which are all not being detected, but this thread is mainly about the keyboard as the K100 RGB is supposed to be supported according to the Wiki.

### Did you try anything that may have helped?

I tried adding the USB Quirk noted in the Troubleshooting page of the Wiki. I unplugged and replugged the keyboard as well as rebooted the Computer.

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-report-1733223645.gz](https://github.com/user-attachments/files/17991736/ckb-next-report-1733223645.gz)

masterdokuro, 2024-12-16:
I'm on fedora (gnome) and the dnf version of ckb-next has never detected my K100 keyboard.  I have always has to build from source using the fedora instructions here https://github.com/ckb-next/ckb-next/wiki/Linux-Installation.  This will give you version **ckb-next 0.6.0-53-g1ce2909** and the keyboard will be detected.

Also, be aware, if your using wayland make sure to read this issue https://github.com/ckb-next/ckb-next/issues/1097.

gravityfargo, 2025-05-13:
Support has been added as of March 2025
https://github.com/ckb-next/ckb-next/releases/tag/v0.6.1

closed 2025-05-13 by gravityfargo