ckb-next/ckb-next #1120: K70 PRO

darqfleet, 2024-12-20:
### What kind of device is it?

Keyboard

### What is the name of the device?

K70 PRO

### Link to Corsair's Product Page

https://www.corsair.com/pl/pl/p/keyboards/ch-9109410-na/k70-rgb-pro-mechanical-gaming-keyboard-with-pbt-double-shot-pro-keycaps-cherry-mx-red-ch-9109410-na?srsltid=AfmBOoolrlOcU3jnt2gRUja5bqnKeYQjnk41MEz2DCSGFqsqUOOuAOHe

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-report-1734700480.gz](https://github.com/user-attachments/files/18211096/ckb-next-report-1734700480.gz)

willasm, 2024-12-28:
There is a pull request that adds this keyboard. I am using it and it is working great.

willasm, 2025-01-04:
Yes I believe you are correct. Mine is K70 RGB PRO.

closed 2025-05-13 by gravityfargo