ckb-next/ckb-next #1122: CORSAIR K70 CORE RGB

Yasha613, 2024-12-28:
### What kind of device is it?

Keyboard

### What is the name of the device?

CORSAIR K70 CORE RGB Mechanical Gaming Keyboard

### Link to Corsair's Product Page

https://www.corsair.com/us/en/p/keyboards/ch-910971e-na/k70-core-rgb-mechanical-gaming-keyboard-black-ch-910971e-na

### Additional Details

noone@fedora:~/ckb-next$ lsusb | grep K70
Bus 003 Device 017: ID 1b1c:1bff Corsair CORSAIR K70 CORE RGB Mechanical Gaming Keyboard

### Generated Report

[ckb-next-report-1735374787.gz](https://github.com/user-attachments/files/18265571/ckb-next-report-1735374787.gz)

AmyMagdalena, 2025-02-17:
[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/18831323/ckb-next-dev-detect-report.gz)

Adding my own detect report to this, I don't think it'll add much though, OS is arch linux, latest zen kernel,

 btw, for anyone that needs to disable the weird pulsing effects, fn+0 sets to rgb solid colour, keep pressing until you get a desirable colour, other built in commands are in the manual

gravityfargo, 2025-05-13:
Duplicate of #1010