ckb-next/ckb-next #1125: M75 Wireless Support

PeterFour, 2025-01-14:
### What kind of device is it?

Mouse

### What is the name of the device?

M75 Wireless

### Link to Corsair's Product Page

https://www.corsair.com/uk/en/p/gaming-mouse/ch-931d010-eu/m75-wireless-lightweight-rgb-gaming-mouse-black-eu-ch-931d010-eu

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/18416975/ckb-next-dev-detect-report.gz)

PeterFour, 2025-01-16:
I'm interested in looking into adding support for it since I have the device. Just don't know how easy that would be as a first task

Betim-Hodza, 2025-05-07:
I'm interested as well, does anyone know what files need to be modified to update the support?

Betim-Hodza, 2025-05-11:
I was working on this for the past week or so on the main git branch, but never actually created a fork until I got it working. It's not perfect, but I tried to copy the settings from the m65 and I got the wired version to actually be detected, working mouse movement, and scrolling. Changeable color profiles, monitor DPI. I haven't been able to fix the bindings, and ik copying from older mice, which could not be 1 to 1, but this is the farthest I've gotten. The more problems I run into, ill try to get them fixed, but I'm happy now that I can control some of the color on my mouse!

https://github.com/Betim-Hodza/ckb-next-m75-support-/tree/master

gravityfargo, 2025-05-13:
Notes: 
- The wireless dongle is the same pid as the M75 Air #1112

zoni, 2026-01-08:
> Would like to investigate wireless as well, but that seems to be more complicated considering the dongle is a generic device (found this conversation: https://github.com/ckb-next/ckb-next/pull/1194 ).

I've been working on wireless support for the M75 and have it largely working, except DPI switching is completely non-working and there's some other sharp edges still. I've not had the motivation to continue work on it for the past couple of weeks, but give it a few more weeks and I'll probably have had a chance to continue (and attempt to rebase my patchset on top of this wired support)