ckb-next/ckb-next #1126: Corsair M65 RGB Ultra Gaming Mouse - 1ba6

marjonz, 2025-01-16:
### What kind of device is it?

Mouse

### What is the name of the device?

Corsair M65 RGB Ultra Gaming Mouse

### Link to Corsair's Product Page

https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/FPS-Fast-Action-Mice/M65-RGB-ULTRA-WIRELESS-Tunable-FPS-Gaming-Mouse/p/CH-9319411-NA2

### Additional Details

Add any additional details here.
`Bus 001 Device 013: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver`

### Generated Report

Drag your generated report here.
[ckb-next-report-1737013515.gz](https://github.com/user-attachments/files/18435128/ckb-next-report-1737013515.gz)

Unexian, 2025-01-26:
Bump bc I also want this

gravityfargo, 2025-05-13:
Does this mouse work solely as a wired device? My Dark Core RGB Pro will work via USB with the toggle under the mouse set to the off position (middle). If so, could you attach another report without the wireless dongle?

Notes:
- related to M65 RGB ULTRA WIRELESS Gaming Mouse - 1bb5 #950 
- related to M65 RGB ULTRA Mouse - 1b9e #912