ckb-next/ckb-next #1127: New Corsair Scimitar RGB Elite not detected

rack-tiger, 2025-01-18:
### Affected Devices

Corsair Scimitar RGB Elite

### Linux Distribution

Kubuntu 24.04

### Installation Method

Ubuntu Software app

### ckb-next version

ckb-next 0.6.0

### What did you expect?

New (2025) Corsair Scimitar RGB Elite not detected. Old Version works.

[image]

### Did you try anything that may have helped?

- Change USB Port.
- Restart PC.

### Additional Details

Add any additional details here.

### Generated Report

[ckb-next-report-1737198576.gz](https://github.com/user-attachments/files/18464180/ckb-next-report-1737198576.gz)

Drag your generated report here.

Yueh72, 2025-03-29:
I got the same issue

nilbest, 2025-04-03:
Me too.
Sadly, this was the only reason why I bought it.
I hoped to get it working on Linux too.

Voidheart, 2025-04-18:
Yep it seems the 2025 model is different, my old one worked

Kronkelman, 2025-05-09:
I'm running into the same issue on EndevourOS. My old Scimitar works fine, the new one is not detected. From what I can tell by the report, my computer seems to recognise the new mouse, but ckb-next does not. I'm attaching the report in case it's useful!

[ckb-next-report-1746806439.gz](https://github.com/user-attachments/files/20125405/ckb-next-report-1746806439.gz)

XorOwl, 2025-05-10:
same here
Mint 22.1
tried both 6.0 installed through apt, and manual compile, but it doesn't detect it

gravityfargo, 2025-05-13:
This device is not currently supported. See
SCIMITAR RGB ELITE Gaming Mouse - 1be3 #1094

closed 2025-05-13 by gravityfargo