ckb-next/ckb-next #1134: Keyboard always starts with brightness 0%

phlepper, 2025-03-06:
### Describe your situation in detail.

I'm not sure if this is a bug or I messed something up, but I have ckb-next in my startup and when it starts, it causes the keyboard to go blank (unlit).  I can then use the brightness top row button and it cycles through the brightnesses correctly with the correct LED animations and colors.  But how do I get it to default to 100% brightness?

I am using a Corsair RGB Strafe keyboard with Pop! OS 22.04 and installed ckb-next with apt.

phlepper

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

tatokis, 2025-03-12:
It sounds like the GUI either can't save your settings (check the permissions in `~/.config/ckb-next`) or you have a buggy old version that would sometimes do that due to "Turn lights off when idle". Try turning that off in the settings.

phlepper, 2025-03-12:
I checked the permissions on the folder and config files within and they were read-write.  I turned off the "Turn lights off when idle" and that fixed the problem.  Thanks!