ckb-next/ckb-next #1135: bindings not working after upgrade to v0.6.1

Nummer42O, 2025-03-10:
### Affected Devices

M55 RGB Pro

### Linux Distribution

Ubuntu 22.04 LTS

### Installation Method

apt

### ckb-next version

0.6.1

### What did you expect?

Before the update everything worked fine. I could change the DPI and use my custom bindings on the right buttons.
After the update those bindings don't work anymore, I can not cycle the DPI and the indicator shows the wrong color.

### Did you try anything that may have helped?

- Unplugging and replugging the mouse.
- Using a non-USB-3.0 port.
- Restarting ckb-next as well as Ubuntu.

### Additional Details

When starting ckb-next from the command line, the following output can be seen:
```
ckb-next 0.6.1
Using "NVIDIA Corporation" "NVIDIA GeForce RTX 2070/PCIe/SSE2"
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/nummer42o/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-42736-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
```
The last two lines pop up repeatedly.

### Generated Report

[ckb-next-report-1741619116.gz](https://github.com/user-attachments/files/19165282/ckb-next-report-1741619116.gz)

tatokis, 2025-03-10:
I have no idea what could cause this. In the logs I do see that 0.6.0 struggled to communicate with your devices as well, so it's possible this is somehow coincidental, or an unrelated change is causing this to happen now.

Can you control the lights on the mouse?

If it's consistent, then you could try to git bisect the issue.

Not sure else what you could try. Maybe the usb quirks in the wiki? A USB hub? Or not using a USB hub if you are using one right now. Using the rear ports directly on your computer (if a desktop), etc etc.

A slightly unrelated question: Did "save to hardware" work at all on your K55? I see it failing to load the hardware profile using the old profile format, and since this is a relatively new keyboard I suspect it doesn't support it.

LuPro, 2025-03-10:
Having the same issue (lights work, but DPI change and button bindings don't - eg: the right side forward/backwards buttons don't work) on the same mouse after the upgrade:

```
ckb-next 0.6.1
Using "AMD" "AMD Radeon RX 7900 XT (radeonsi, navi31, LLVM 19.1.7, DRM 3.61, 6.13.5-2-MANJARO)"
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/luis/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-147224-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
Scanning "/usr/local/libexec/ckb-next-animations/heat"
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
Scanning "/usr/local/libexec/ckb-next-animations/life"
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
Scanning "/usr/local/libexec/ckb-next-animations/rain"
Scanning "/usr/local/libexec/ckb-next-animations/random"
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
Scanning "/usr/local/libexec/ckb-next-animations/snake"
Scanning "/usr/local/libexec/ckb-next-animations/wave"
Starting "/usr/local/libexec/ckb-next-animations/heat"
Creating PulseAudio context
```

[ckb-next-report-1741647976.gz](https://github.com/user-attachments/files/19171857/ckb-next-report-1741647976.gz)

Installed by pulling the repo, checking out the release tag and then `./quickinstall`.

Happy to debug stuff, but would need a pointer what to test.

Nummer42O, 2025-03-11:
> Can you control the lights on the mouse?

Yes. For the "normal" light as well as the DPI indicator.

> If it's consistent, then you could try to git bisect the issue.

I'm not sure how long it will take me but I will try to do that.

> Not sure else what you could try. […]

I am on the rear ports by default. I tried the front I/O today but that didn't help. I don't really have another USB Hub I could use although if this is a likely solution I could probably borrow one.

> A slightly unrelated question: […]

I'm not 100% sure what you mean by that? If you mean that the Keyboard retains information about the color profile even when ckb-next is not running, then no - it does not do that. The mouse seems to keep that information (at least regarding the DPI) but they keyboard doesn't.

Another little fun fact I found while trying stuff: After trying to rebind the dpi key it would set me on my lowest DPI setting without a way to change it. After restarting the computer I had my default DPI back (while still displaying the wrong color in the indicator).

tatokis, 2025-03-12:
> I'm not 100% sure what you mean by that? If you mean that the Keyboard retains information about the color profile even when ckb-next is not running, then no - it does not do that. The mouse seems to keep that information (at least regarding the DPI) but they keyboard doesn't.

There should be a button on the bottom left that says "save to hardware". Does pressing it actually make the settings stick?

> I am on the rear ports by default. I tried the front I/O today but that didn't help. I don't really have another USB Hub I could use although if this is a likely solution I could probably borrow one.

Since more people are encountering this, then it's unlikely a connectivity issue. I'll have another look to try to identify it. I haven't been able to reproduce this on my end (and I've been using the latest available code every day). It seems to be M55 specific.

A git bisect would make this much easier. Thanks, and apologies for this.

Nummer42O, 2025-03-12:
No worries about it - I'm just happy that this repository exists!

Doing the git bisect was a bit tricky but I got it down to this commit:
```
ff88b9bd7ebbef101c7667b96a20d7ce22ddf49f is the first bad commit
commit ff88b9bd7ebbef101c7667b96a20d7ce22ddf49f
Author: Tasos Sahanidis <tasos@tasossah.com>
Date:   Thu Aug 11 05:54:37 2022 +0300

    Add K100 to GUI and daemon
    
    Readd alt K100 lighting method
    
    Do not send 2nd lighting data
    
    It is moved to init because I can't easily tell when CUE sends it,
    and it's definitely not once every frame.
    
    Additionally, it is commented out because it broke some lights on certain K100s
    
    Fix K100 control wheel in GUI
```

> Does pressing it actually make the settings stick?

That would require me to be able to click it. ^^

[image]

tatokis, 2025-03-13:
Can you please try the latest code from git? (Or apply https://github.com/ckb-next/ckb-next/commit/7a7d0c80bf1ef56a024606ae49551ce84872ce72 manually.)

What a mess.

Nummer42O, 2025-03-17:
Since the commit fixes it and noone had any complains with it I think we can claim this as done.

closed 2025-03-17 by Nummer42O