ckb-next/ckb-next #1139: Application fails to run on Arch Linux due to libquazip update

armv8-a, 2025-03-18:
### Affected Devices

N/A

### Linux Distribution

Arch

### Installation Method

ckb-next and ckb-next-git (AUR)

### ckb-next version

0.6.2.r1.gc70a4ae (ckb-next-git pkgver)

### What did you expect?

The application should start.

### Did you try anything that may have helped?

Tried building with `-DPREFER_QT6=ON`. Does not work.
Tried fooling it with `# ln -s /usr/lib/libquazip1-qt5.so.1.5.0 /usr/lib/libquazip1-qt5.so.1.4.0`. This works, so perhaps it just needs a version bump.

### Additional Details

```sh
$ pacman -Q quazip-qt{5,6}
quazip-qt5 1.5-1
quazip-qt6 1.5-1

$ ckb-next
ckb-next: error while loading shared libraries: libquazip1-qt5.so.1.4.0: cannot open shared object file: No such file or directory
```

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/19304359/ckb-next-dev-detect-report.gz)

tatokis, 2025-03-18:
https://wiki.archlinux.org/title/Arch_User_Repository#Updating_packages

You are not actually rebuilding the package if you are still getting this error. It's impossible to link against a library that no longer exists.

tatokis, 2025-03-18:
You are also not running ckb-next from the AUR.

closed 2025-03-18 by tatokis

armv8-a, 2025-03-19:
> You are also not running ckb-next from the AUR.

Oh my. You're absolutely right; I forgot I had manually installed an older version in /usr/local. That was dumb. Sorry!