ckb-next/ckb-next #1140: Ckb-next update now not recognizing mouse.

Dynasty8614, 2025-03-18:
### Affected Devices

Corsair IronClaw RGB

### Linux Distribution

Fedora 41

### Installation Method

RPM

### ckb-next version

.6.1

### What did you expect?

I got on my computer today and updated everything including ckb-next. However after I updated, it turned off experimental mode, so I turned it back o, but my Ironclaw mouse is now not being recognized, my side buttons do not work and the DPI I set it at has now changed.

### Additional Details

Add any additional details here.

[image]

### Generated Report

Drag your generated

 report here.

Dynasty8614, 2025-03-18:
[ckb-next-report-1742317359.gz](https://github.com/user-attachments/files/19323886/ckb-next-report-1742317359.gz)

Not sure why my generated report didnt paste

gravityfargo, 2025-05-13:
Please generate a report and attach it to a new issue.
https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#generate-a-report

closed 2025-05-13 by gravityfargo