ckb-next/ckb-next #1141: Restore from Tray does not show window

Grim-GH, 2025-03-19:
### Affected Devices

K95 RGB PLATINUM, NIGHTSWORD RGB

### Linux Distribution

CachyOS

### Installation Method

yay --rebuildall -S ckb-next-git

### ckb-next version

AUR Explicit (1): ckb-next-git-1:0.6.0.r0.gfaca39b1-1

### What did you expect?

After system update got an error that indicated that ckb needed to be rebuilt. 
Rebuilt it using `AUR Explicit (1): ckb-next-git-1:0.6.0.r0.gfaca39b1-1`.
After install tested again and same issue. When I click the "x" on the window to close it, ckb stays in the tray, but when I try to right click and 'Restore" from the Tray icon I get the ckb window title in the applciation bar, but the window never shows. 
KDE Plasma issue maybe? 
Was doing the same thing with the non-git install prior to System update.

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.

Grim-GH, 2025-03-26:
Rebooting has not made a difference. This has been like this between reboots, system updates, ckb-reinstall from git after system update, and even between system sleeps.
Re-run tries to run it (Hourglass cursor), then it just doesn't do anything (Force quit I believe of second run).

Don't know if this makes a difference either, but I'm using KDE and when I click "Restore" I get 2 "ckb-next" applications showing up in the application panels (Taskbar) on each monitor (Dual monitors).

tatokis, 2025-03-26:
I suspect it might be a Plasma regression, especially if it started occurring after a system update and not a ckb-next update. I'm not really sure. Can you try a few other applications? Perhaps try building the GUI with Qt6. (Unsure what the dependencies are on Arch, but you pass -DPREFER_QT6=1 to cmake and then ensure it says "Building with Qt6", and the About Qt button says Qt 6.)

Grim-GH, 2025-03-27:
I don't have the any problem with OpenRGB, Steam, or Discord which all have tray icons.