ckb-next/ckb-next #1144: Inclusion or Work-around for special characters

joshuactk, 2025-03-24:
### Enhancement Details

Context: I have a Japanese keyboard, but I want to enable my keyboard to write in French using the CSA layout, which is QWRTY-friendly. With my present layout, the only button that is missing is the one for the character 'ù'.

Sharing the layout for your reference:
[image]
[image]

Feature Request: 

1. Enable a way to use say macros to assign a special character. Currently, the keys that can be rebound must exist in the layout of the keyboard on a hardware-level.
2. Enable a way for keys with new binding to change their output when pressing modifiers like 'Shift' or 'Alt-Gr'.

### Additional Details

Add any additional details here.

tatokis, 2025-03-24:
ckb-next works with physical keys (scancodes) and not virtual keycodes. Adding that would mean performing desktop-specific key injection that tends to be wonky (especially when games use raw input).

I suggest binding a key to run xte (i.e. `xte 'str αβγ'`) if you want to type specific characters under X11. No idea about Wayland.

joshuactk, 2025-09-09:
For anyone with a similar problem, I used ckb-next to remap the B00 button in the ISO layout to F24 (anything within F13-F24 works), then installed [input-remapper](https://github.com/sezanzeb/input-remapper) to convert F24 to the 'less' button (standing for <> in the ISO/IEC 9995-1/2 layout) as a workaround to the issue.