ckb-next/ckb-next #1146: Install Error on Bazzite

MarcTheCyborg, 2025-03-29:
### Describe your situation in detail.

I'm fairly new to Linux, so this might have a simple fix, but when i use the rpm-ostree install gh repo clone ckb-next/ckb-next or rpm-ostree install https://github.com/ckb-next/ckb-next.git commands, it seems to run through an installation project before giving me a package not found error. Am I missing something? 

[image]

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

DeVelox, 2025-04-05:
You can think of `rpm-ostree` as a package manager, meaning the command you would want to use is simply `rpm-ostree install ckb-next`.

It doesn’t understand GitHub links, but in this case ckb-next is already provided by the Fedora repositories which Bazzite is based on. 

For more information on how to install software on Bazzite refer to https://docs.bazzite.gg/Installing_and_Managing_Software/

gravityfargo, 2025-05-13:
> ckb-next is already provided by the Fedora repositories

I'm not a fedora user, but their docs are very informative. https://docs.fedoraproject.org/en-US/quick-docs/finding-and-installing-linux-applications/

I image the command line version would be 
```bash
dnf install ckb-next
```

closed 2025-05-13 by gravityfargo