ckb-next/ckb-next #1150: Back mouse button not working

BrownyPlaysRoblox, 2025-04-02:
### Affected Devices

Corsair Nightsword RGB Gaming Mouse

### Linux Distribution

Arch Linux

### Installation Method

AUR

### ckb-next version

ckb-next 0.6.2

### What did you expect?

If I press this down button (circled in image) on my mouse, it just does not work... All of the other buttons do work though, and in Windows, the back button also does work. I try to use the back button to activate a macro and it does not work, although if I try the same thing with the rest of the buttons, those buttons work perfectly fine.

[image]

### Did you try anything that may have helped?

Used a different Linux distro (Bazzite) and still had the same issue; unplugging and replugging back in and restarting the computer has not fixed the issue.

### Additional Details

### Generated Report

[ckb-next-report-1743634326.gz](https://github.com/user-attachments/files/19576893/ckb-next-report-1743634326.gz)

LZeugirdor, 2025-09-26:
+1 also having this issue, as well as rebinding it not working.

BrownyPlaysRoblox, 2025-11-09:
Yes still having the same issue.