ckb-next/ckb-next #1158: CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard 1baf - Conflicting information - Supported or not?

mikeybob, 2025-05-26:
### What kind of device is it?

Keyboard

### What is the name of the device?

CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard

### Link to Corsair's Product Page

https://www.corsair.com/uk/en/p/keyboards/ch-9194010-uk/k65-rgb-mini-60-mechanical-gaming-keyboard-cherry-mx-red-uk-black-ch-9194010-uk

### Additional Details

This is in the wiki as supported in ckb-next 0.6.1 but I'm currently running version 0.6.2 and the keyboard is not supported. Just curious which is correct. 

- Issue #721 was recently closed without further information. 
- Commit bd76bae referenced in the closing comment is from Jun 23, 2023.

### Generated Report

[ckb-next-report-1748232705.gz](https://github.com/user-attachments/files/20434770/ckb-next-report-1748232705.gz)

tatokis, 2025-05-26:
I'm confused. It was supposed to be confirmed working. The 1baf is in the list, yet the daemon doesn't even make an attempt to open it according to your report.

mikeybob, 2025-05-26:
Ran with the daemon experimental flag enabled, and unfortunately the keyboard does not function and does not appear in the ui. This seems to be the same state reported on September 14, 2024 in #721.

tatokis, 2025-05-27:
Can you please send me the output of the daemon when ran manually from a terminal with experimental devices enabled?

mikeybob, 2025-06-20:
I have decided to use a different solution.

closed 2025-06-20 by mikeybob