ckb-next/ckb-next #1160: Macro buttons cause held inputs to be cleared/dropped

the-fifth, 2025-06-04:
### Affected Devices

K95 Vengeance

### Linux Distribution

Mint

### Installation Method

System Package (Software)

### ckb-next version

ckb-next 0.6.2

### What did you expect?

Currently pressing a macro key (any) assigned to a 'typing' method of '/' from numpad (or other key) and also as a 'macro' method including the keydown+keyrelease event causes any held keys to be cleared or dropped, requiring you to release and then press again to have them register.

Example:
Holding the R key, typing a continuous stream of 'r' characters into a text file, pressing G9 which is assigned the action of 'Numpad /' causes a single / to be output to the text file but the R key is no longer outputting to the text file and must be released and depressed again to resume normal behaviour.

### Did you try anything that may have helped?

Have tried non numpad keys, regular keys, irregular/nonstandard (F13+ etc) keys, macro specifically containing keydown/keyup events.

### Generated Report

[ckb-next-report-1749018300.gz](https://github.com/user-attachments/files/20586066/ckb-next-report-1749018300.gz)

the-fifth, 2025-06-05:
I have tried with both macros (key down, default duration/manually setting duration, key up etc)

A further example that would probably highlight it better:
- In a game, fullscreen or otherwise (not sure if that will matter, but best to be thorough at least), binding any of the G keys to NUM / will result in any keys held (ie movement/WASD keys) being cleared if moving, meaning you again need to release the WASD key and repress to trigger. If I press the actual NUM /, the input from WASD isn't dropped. This is reproducible with any keybind, not just numpad keys etc

tatokis, 2025-06-05:
Can you verify if this only happens with the G keys? Unbind one of the G keys completely and press it. Does that let go of WASD? In theory it should do nothing.

the-fifth, 2025-06-05:
I can confirm that pressing _any_ of the G keys causes WASD input to stop, that's interesting I'd never thought to try to test if it was just the G keys themselves.

tatokis, 2025-06-07:
I [have a suspicion](https://github.com/ckb-next/ckb-next/blob/ce55984bd9d0abfbe7b5d3a598923f9b49342ac9/src/daemon/keymap.c#L947-L951) as to what might be at fault, and while I came up with a possible solution, I need more information as I am not familiar with how the legacy devices actually send keys to the computer.

Can you please follow the [USB Capture instructions](https://github.com/ckb-next/ckb-next/wiki/USB-Capture-Setup#wireshark-usb-capture) and generate a capture doing the following?

1. `sudo systemctl stop ckb-next-daemon`
2. Start capture in Wireshark
3. Press and release the following keys in the following order
     1. <kbd>G1</kbd>
     2. <kbd>G18</kbd>
     3. Left <kbd>Shift</kbd>
     4. <kbd>W</kbd>
     5. <kbd>↑</kbd>
4. Stop the Wireshark capture and save it as "K95_Legacy_hw"
5. `sudo systemctl start ckb-next-daemon`
6. Wait at least 2 minutes (this is [due to a bug](https://github.com/ckb-next/ckb-next/commit/d6a9b7a594ed3845f2715bcc631a4b78ede616a8) that has just been fixed, you should consider adding the ckb-next-git PPA afterwards)
7. Start a new Wireshark capture
8. Repeat step 3
9. Stop the Wireshark capture and save it as "K95_Legacy_sw"
10. Upload both here

the-fifth, 2025-06-07:
I will look at adding the PPA directly, unfortunately I am unable to add the .pcapng files directly to the GH issue, is there a preferred host I can link these from? I'm semi-confident there is no overtly sensitive data, just the keycombos and some mouse movement to pause the capture but still.

the-fifth, 2025-06-07:
[K95_Legacy_hw.pcapng.gz](https://github.com/user-attachments/files/20637301/K95_Legacy_hw.pcapng.gz)
[K95_Legacy_sw.pcapng.gz](https://github.com/user-attachments/files/20637300/K95_Legacy_sw.pcapng.gz)

Thanks for pointing that out, I only attempted to .7z them, I've attached the files.

the-fifth, 2025-06-08:
I can confirm that the functionality works, if I test in a game I can keep moving with WASD while activating G keys, macro/direct key etc

Did you want me to repeat the capture test as per the previous method?

commit 067dbb9d (2025-06-08): Merge branch 'legacy-6kro-gkey-fix'

Closes #1160