ckb-next/ckb-next #1161: Vengeance K90 Keyboard "No devices connected"

RoninTech, 2025-06-05:
### Affected Devices

Vengeance K90 Keyboard

### Linux Distribution

Debian Bullseye on Raspberry pi

### Installation Method

Built from source

### ckb-next version

ckb-next 0.6.2-23-g9739675

### What did you expect?

Based on the README it should be supported/detected:

K90 	VENGEANCE 	Supported 	1b02 	≤3.0

### Additional Details

I would like to create some macros with ckb-next but aren't sure if the K90 should be detected or not?  The logs list the K90 as detected but for some reason ckb-next doesn't.

### Generated Report

[ckb-next-report-1749140558.gz](https://github.com/user-attachments/files/20614502/ckb-next-report-1749140558.gz)

RoninTech, 2025-06-05:
Thanks for the prompt reply!  Here you go:

```
me@pi5:~/ $ ckb-next
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
ckb-next 0.6.2-23-g9739675
Using "Broadcom" "V3D 7.1.7.0"
Path to settings: "/home/me/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-715178-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
Scanning "/usr/local/libexec/ckb-next-animations/heat"
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
Scanning "/usr/local/libexec/ckb-next-animations/life"
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
Scanning "/usr/local/libexec/ckb-next-animations/rain"
Scanning "/usr/local/libexec/ckb-next-animations/random"
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
Scanning "/usr/local/libexec/ckb-next-animations/snake"
Scanning "/usr/local/libexec/ckb-next-animations/wave"
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1
could not find valid model information: "" produced -1

```

RoninTech, 2025-06-05:
Sure enough, when I change /dev/input/ckb1

From: corsair  bind notify hwload
To: corsair K95L bind notify hwload

ckb-next seems to now detect the keyboard.

What process populates /dev/input/ckb1?

tatokis, 2025-06-05:
Awesome. Cheers!

commit 366ff14e (2025-06-05): Merge branch 'fix-k90-legacy'

Closes #1161