ckb-next/ckb-next #1165: ./quickinstall not working

J3DP1, 2025-06-17:
### Describe your situation in detail.

When I first try to useit says permission denied so from there I use cmod +x and then after that I type in ls which shows me the files. I try to use ./quickinstall
and I get this ---
./quickinstall: line 51: cmake: command not found
An error occurred,
press enter to exit.
Im new to this so I am very confused!

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

tatokis, 2025-06-18:
You are attempting to build from source. This process requires other packages (dependencies) in order to build it. You can find more information in the installation section https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source

However, it sounds like you should probably not be building from source. Depending on your distribution, you can install a package that is provided, or pick one from here https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#recommended-packages

J3DP1, 2025-06-18:
witch package should I use? I am on cromebook linux.

closed 2025-06-23 by J3DP1