ckb-next/ckb-next #1171: is there a reason why N_KEYS_EXTENDED is not dynamic?

mmtmn, 2025-06-29:
### Describe your situation in detail.

to fix this it would require a substantial refactoring, I'm asking this before putting time into refactoring this. If there is a reason for it (such as firmware blocks such) I'd not touch it.

any insights into this are highly welcomed.

location in repo: (src/daemon/keymap(c/h))

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

Ravenslofty, 2025-06-29:
Sorry, what do you mean by "dynamic" here?

mmtmn, 2025-06-29:
@Ravenslofty what i meant is that N_KEYS_EXTENDED isn’t dynamic, it’s just a constant that gets set at compile time. it doesn’t change depending on what keyboard you plug in or how many keys the device actually has. if the keyboard has more or fewer keys, the code still uses whatever number was hardcoded. so it’s not calculated or adjusted during runtime.

Ravenslofty, 2025-06-29:
before I answer that, can you be honest with me? are you using AI for your pull request?

mmtmn, 2025-06-29:
also, are you talking about #1170? someone gave up on this repo after the nasty bug over at k100, which really seems like an overflow issue

Ravenslofty, 2025-06-30:
If there are enough or too many keys in the keymap, this is effectively harmless; if there are not enough keys, there is a more fundamental issue of understanding that will not be resolved by making N_KEYS_EXTENDED a variable.

By the way, that question was asked in good faith, and my reward was waking up to a bunch of angry pings on IRC. I wasn't having a bad day, but now I surely am.

mmtmn, 2025-06-30:
I see, thank you for the explanation. There's probably an overflow happening somewhere.

Sorry to hear you are having a bad day now. I'd personally never approach a question like that, specially from a new comer trying to understand and help out on a decade old repo that leaked a user's password (zDEFz mentioned this was the reason why he stopped using ckb-next). For [example](https://github.com/openai/whisper/pull/2549) this is how I'd approach something many times worse. I still don't see the purpose behind your question or why you'd even consider it to be in good faith, but whatever.

Regardless, listen, I'll try to debug this further next Sunday, I'll see if there is any overflows or underflows and if nothing too obvious pops up I'll see if i can find it with gdb and ghidra. Along the way, I might ask some structural questions on what seems odd to me. Other contributors mentioned nothing like this has happened to them so far so I'll continue to help as long as we can get straight to the point instead of wasting our times with non-related stuff.

tatokis, 2025-07-01:
> also, are you talking about https://github.com/ckb-next/ckb-next/pull/1170? someone gave up on this repo after the nasty bug over at k100, which really seems like an overflow issue

> to understand and help out on a decade old repo that leaked a user's password (zDEFz mentioned this was the reason why he stopped using ckb-next).

Please stop spreading misinformation. https://github.com/ckb-next/ckb-next/pull/1170#issuecomment-3016790475

mmtmn, 2025-07-01:
Anyways, the question i had has been solved on my end and I don't see how it can be productive to not close this.

closed 2025-07-01 by mmtmn

mmtmn, 2025-07-01:
fuck, i shouldn't be looking at this after work, i really want to fix this hahah if no one solves it before sunday I'll give it another go

mmtmn, 2025-07-13:
@tatokis just adding to the collection of things that caused trouble because of this issue:

https://github.com/ckb-next/ckb-next/issues/1037#issuecomment-2733584957