ckb-next/ckb-next #1174: Is there any work on dark theme?

mmtmn, 2025-07-02:
### Enhancement Details

Hi, this is a quick question, has anyone worked or is currently working on a dark theme for the graphical user interface? If any previous attempts have been conducted, I'd appreciate the lore behind it.

I think the question label would fit this "issue".

### Additional Details

Add any additional details here.

josephcrowell, 2025-07-02:
It's a Qt based interface, so setting a Qt/KDE theme should switch it.

[image]

mmtmn, 2025-07-02:
Interesting, seems like it was built with KDE in mind but it doesn't seem to have a scalable solution to it

mmtmn, 2025-07-02:
thanks @josephcrowell, this helps understand this better.

closed 2025-07-02 by mmtmn