ckb-next/ckb-next #1189: GUI will no longer render on Arch with Hyprland (Wayland)

zabulonian, 2025-07-30:
### Affected Devices

n/a

### Linux Distribution

Arch

### Installation Method

AUR

### ckb-next version

ckb-next-git in AUR, currently: ckb-next 0.6.2-48-g55ea82e0 

### What did you expect?

Starting around the end of May or so, I became unable to pull up the GUI.
Using Hyprland (so, in Wayland), and have an Nvidia card. However, testing has revealed it is likely not a graphics driver/Nvidia issue (see below)

ckb-next is set to launch automatically, but even when quitting and launching manually, the GUI won't come up. Right clicking the tray icon and clicking 'Restore' doesn't do anything; though regardless of launch method it does successfully apply the custom animation I'd previously created.

Initially, when first reporting this issue, this is what would print to my terminal when launching that way:

```
ckb-next 0.6.2-40-gb2fa1708
Using "NVIDIA Corporation" "NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2"
Path to settings: "/home/[user]/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-3962-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
QOpenGLContext::makeCurrent() called with non-opengl surface 0x55b197339c90
QRhiGles2: Failed to make context current. Expect bad things to happen.
Failed to create QRhi for QBackingStoreRhiSupport
QOpenGLContext::makeCurrent() called with non-opengl surface 0x55b197339f70
QRhiGles2: Failed to make context current. Expect bad things to happen.
Failed to create QRhi for QBackingStoreRhiSupport
Starting "/usr/lib/ckb-next-animations/gradient"
Creating PulseAudio context
```

At the time, I did not see any additional output in the terminal when trying to bring up the GUI with right click -> restore

Issue has persisted since then, across multiple updates, restarts, etc. 

### Did you try anything that may have helped?

With tatokis' guidance, I tried launching ckb next as follows:
```__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json LIBGL_ALWAYS_SOFTWARE=1 ckb-next```

This produced similar results:
```
Limiting MSAA to 0 because of "Mesa" "llvmpipe (LLVM 20.1.8, 256 bits)"
ckb-next 0.6.2-48-g55ea82e0
Using "Mesa" "llvmpipe (LLVM 20.1.8, 256 bits)"
Path to settings: "/home/mossy/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-70682-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
QOpenGLContext::makeCurrent() called with non-opengl surface 0x55c4e81e6f60
QRhiGles2: Failed to make context current. Expect bad things to happen.
Failed to create QRhi for QBackingStoreRhiSupport
QOpenGLContext::makeCurrent() called with non-opengl surface 0x55c4e81e7240
QRhiGles2: Failed to make context current. Expect bad things to happen.
Failed to create QRhi for QBackingStoreRhiSupport
Starting "/usr/lib/ckb-next-animations/gradient"
Creating PulseAudio context
QOpenGLWidget: No fbo, cannot render
QOpenGLWidget: No fbo, cannot render
QOpenGLWidget: No fbo, cannot render
```

Those 3 new lines at the bottom begin to print infinitely when I try to pull up the GUI by clicking on the tray icon repeatedly. This happens regardless of the method of launching ckb-next. 

### Generated Report

(Unsure this is actually needed as it is not related directly to a device, but here it is just in case)
[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/21516935/ckb-next-dev-detect-report.gz)

tatokis, 2025-07-31:
Hi!

The report was actually useful since I saw you have a few Qt-related env vars set.
Can you try unsetting all of them and trying again with llvmpipe?

```
printenv | grep QT
```

tatokis, 2025-08-01:
I did test with qt6ct when trying to reproduce this, but it did work for me. Could it perhaps be some configuration setting in qt6ct? What style do you have set in it? It's possibly a bug with a specific style.

Try renaming ~/.config/qt6ct to something else and test again perhaps.

zabulonian, 2025-08-01:
Renamed the folder and yep, it's something in ~/.config/qt6ct

I've zipped up the folder and attached it below. I know I'm using kvantum, but I couldn't tell you further intricacies; my current setup is based on HyDE from a couple of months ago; I haven't updated HyDE since installing and have made my own tweaks, but none to qt6ct or kvantum. 
ckb-next's GUI did work originally with this setup, though. 

[qt6ct.zip](https://github.com/user-attachments/files/21555105/qt6ct.zip)