ckb-next/ckb-next #1203: sysvinit ckb-next-daemon.sh "reload" command hangs driver instead of reloading the configuration

spillner, 2025-08-31:
### Affected Devices

STRAFE (non-RGB)

### Linux Distribution

Slackware

### Installation Method

From git

### ckb-next version

0.6.2-70-g02c2f2df

### What did you expect?

The "reload" command should prompt the daemon to reload its configuration files, and possibly to reset the status of its connection with any supported hardware devices.  Instead, it causes the driver to immediately hang, because the `ckb-next-daemon.sh` script is set up to set SIGUSR1 to the running `ckb-next-daemon` process, but that signal is not handled by the daemon code.

### Did you try anything that may have helped?

Immediately upon issuing the `reload` command, the driver hangs, rendering the keyboard unusable.  The keyboard can be reactivated by disconnecting and reconnecting it to the USB port.

### Additional Details

I'm not sure why `linux/sysvinit/ckb-next-daemon.in` is set up to send SIGUSR1 to the daemon process, since there does not appear to be any corresponding handler for SIGUSR1 in the daemon code.  `src/daemon/main.c` sets up handlers only for SIGTERM, SIGINT, SIGQUIT, and SIGUSR2, with SIGUSR2 doing nothing.

### Generated Report

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/22069216/ckb-next-dev-detect-report.gz)

Ravenslofty, 2025-08-31:
My knowledge of the daemon internals is a bit outdated, but last time I hacked on it, there were no configuration files for the daemon to reload. So... if anything, I think the actual issue is that the sysvinit script has a reload command to begin with.

spillner, 2025-09-01:
That was my impression too--- looks like it writes a few files under `/var/run/ckb`, but doesn't read anything other than its own PID.  But I don't know enough about the driver or its history to guess whether there used to be a purpose behind the "reload" command, or it was just unthinkingly carried over from some other piece of software.  For example, it might make sense to scan for newly connected devices, but it looks like `usbmain()` already does that continuously.