ckb-next/ckb-next #1216: ckb-next crashes in the background, Address boundary error

Frankkz, 2025-09-25:
### Affected Devices

K95 RGB Platinum

### Linux Distribution

Manjaro Linux

### Installation Method

AUR

### ckb-next version

0.6.2.r70.g02c2f2df-1

### What did you expect?

ckb-next should keep running but it crashes after a certain amount time.
The daemon keeps running, but my settings(idle timer & keyboard lights) don't work anymore until I restart the GUI.

### Did you try anything that may have helped?

Restarting manually solves it until it crashes again.

### Additional Details

`$ ckb-next --help-all`
```
ckb-next 0.6.2-70-g02c2f2df
Using "AMD" "AMD Radeon RX 7900 XT (radeonsi, navi31, LLVM 20.1.8, DRM 3.63, 6.15.11-2-MANJARO)"
Path to settings: "/home/user/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-875481-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
Starting "/usr/lib/ckb-next-animations/mviz"
Creating PulseAudio context
fish: Job 1, 'ckb-next --help-all' terminated by signal SIGSEGV (Address boundary error)
```
`$ coredumpctl dump`
```
           PID: 875481 (ckb-next.real)
           UID: 1000 (user)
           GID: 1000 (user)
        Signal: 11 (SEGV)
     Timestamp: Thu 2025-09-25 14:35:07 CEST (8min ago)
  Command Line: /usr/bin/ckb-next --help-all
    Executable: /usr/bin/ckb-next.real
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-org.kde.konsole@68306b4895d44fbf8a0d04cd1fd62b28.service
          Unit: user@1000.service
     User Unit: app-org.kde.konsole@68306b4895d44fbf8a0d04cd1fd62b28.service
         Slice: user-1000.slice
     Owner UID: 1000 (user)
       Boot ID: ee643678cf094e24bfe493d1b62e5b75
    Machine ID: f7e0cfa0e1cc42f1ad1e8a567a167f3c
      Hostname: manjaro
       Storage: /var/lib/systemd/coredump/core.ckb-next\x2ereal.1000.ee643678cf094e24bfe493d1b62e5b75.875481.1758803707000000.zst (present)
  Size on Disk: 5.4M
       Message: Process 875481 (ckb-next.real) of user 1000 dumped core.
[... 314 lines cut ...]
                #1  0x00007f36512931ac n/a (libc.so.6 + 0x931ac)
                #2  0x00007f365129381c n/a (libc.so.6 + 0x9381c)
                #3  0x00007f3651295ede pthread_cond_wait (libc.so.6 + 0x95ede)
                #4  0x00007f363d96455e n/a (libgallium-25.2.1-arch1.4.so + 0x56455e)
                #5  0x00007f363d91de7d n/a (libgallium-25.2.1-arch1.4.so + 0x51de7d)
                #6  0x00007f363d96448d n/a (libgallium-25.2.1-arch1.4.so + 0x56448d)
                #7  0x00007f36512969cb n/a (libc.so.6 + 0x969cb)
                #8  0x00007f365131aa0c n/a (libc.so.6 + 0x11aa0c)
                
                Stack trace of thread 875733:
                #0  0x00007f365129f042 n/a (libc.so.6 + 0x9f042)
                #1  0x00007f36512931ac n/a (libc.so.6 + 0x931ac)
                #2  0x00007f365129381c n/a (libc.so.6 + 0x9381c)
                #3  0x00007f3651295ede pthread_cond_wait (libc.so.6 + 0x95ede)
                #4  0x00007f363d96455e n/a (libgallium-25.2.1-arch1.4.so + 0x56455e)
                #5  0x00007f363d91de7d n/a (libgallium-25.2.1-arch1.4.so + 0x51de7d)
                #6  0x00007f363d96448d n/a (libgallium-25.2.1-arch1.4.so + 0x56448d)
                #7  0x00007f36512969cb n/a (libc.so.6 + 0x969cb)
                #8  0x00007f365131aa0c n/a (libc.so.6 + 0x11aa0c)
                ELF object binary architecture: AMD x86-64
```

### Generated Report

[ckb-next-report-1758804970.gz](https://github.com/user-attachments/files/22537296/ckb-next-report-1758804970.gz)

Frankkz, 2025-09-26:
Not sure if this adds anything useful to you but might as well share; 

`$ journalctl -b | grep ckb-next`
```
sep 26 13:23:04 manjaro kernel: traps: ckb-next.real[1268650] general protection fault ip:7f0120111ac3 sp:7fffa10b82d0 error:0 in libffi.so.8.2.0[7ac3,7f012010c000+7000]
sep 26 13:23:04 manjaro systemd-coredump[1276811]: Process 1268650 (ckb-next.real) of user 1000 terminated abnormally with signal 11/SEGV, processing...
sep 26 13:23:04 manjaro systemd-coredump[1276820]: Process 1268650 (ckb-next.real) of user 1000 dumped core.
                                                   #18 0x000055c81599d03c main (/usr/bin/ckb-next.real + 0x5903c)
                                                   #21 0x000055c81599e4e5 n/a (/usr/bin/ckb-next.real + 0x5a4e5)
                                                   #4  0x000055c815a9b123 n/a (/usr/bin/ckb-next.real + 0x157123)
                                                   #11 0x000055c815a0dc0f n/a (/usr/bin/ckb-next.real + 0xc9c0f)

```

`$ journalctl -b | grep -i wayland`
```
sep 26 13:23:04 manjaro msm_kde_notifier[1806]: qt.qpa.wayland: Creating a fake screen in order for Qt not to crash
sep 26 13:23:04 manjaro kded6[1558]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro xdg-desktop-portal-kde[1656]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro kactivitymanagerd[1651]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro pamac-tray-plasma[1842]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro dolphin[1269874]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro polkit-kde-authentication-agent-1[1654]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro konsole[1268114]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro plasmashell[1600]: qt.qpa.wayland: There are no outputs - creating placeholder screen
sep 26 13:23:04 manjaro org_kde_powerdevil[1655]: qt.qpa.wayland: There are no outputs - creating placeholder screen
                                                   #3  0x00007f0120d0a48d n/a (libwayland-client.so.0 + 0x448d)
                                                   #4  0x00007f0120d0b2e9 n/a (libwayland-client.so.0 + 0x52e9)
                                                   #5  0x00007f0120d0b6f3 wl_display_dispatch_queue_pending (libwayland-client.so.0 + 0x56f3)
                                                   #6  0x00007f0120c74266 _ZN15QtWaylandClient15QWaylandDisplay13flushRequestsEv (libQt6WaylandClient.so.6 + 0x6e266)
                                                   #5  0x00007f0120c6b439 n/a (libQt6WaylandClient.so.6 + 0x65439)
                                                   #4  0x00007f0120c6b4a3 n/a (libQt6WaylandClient.so.6 + 0x654a3)
```

Frankkz, 2025-10-20:
> can you try with `-DUSE_WAYLAND=ON` on a fresh -git AUR build?

cloned fresh aur git and rebuilt with -DUSE_WAYLAND=ON

`$ makepkg -si`
[terminal-build.log](https://github.com/user-attachments/files/23000988/terminal-build.log)

`$ ckb-next --help-all`
```
ckb-next --help-all
ckb-next 0.6.2-75-g4bf942db
Using "AMD" "AMD Radeon RX 7900 XT (radeonsi, navi31, LLVM 20.1.8, DRM 3.64, 6.17.1-0-MANJARO)"
Path to settings: "/home/user/.config/ckb-next/ckb-next.conf"
service is "org.kde.StatusNotifierItem-43418-1"
Registering a client interface to the KStatusNotifierWatcher
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
Starting "/usr/lib/ckb-next-animations/mviz"
Creating PulseAudio context
fish: Job 1, 'ckb-next --help-all' terminated by signal SIGSEGV (Address boundary error)
⏎                      
```

`$ coredumpctl gdb`
```
           PID: 43418 (ckb-next.real)
           UID: 1000 (user)
           GID: 1000 (user)
        Signal: 11 (SEGV)
     Timestamp: Mon 2025-10-20 14:25:48 CEST (21s ago)
  Command Line: /usr/bin/ckb-next --help-all
    Executable: /usr/bin/ckb-next.real
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-org.kde.konsole@d0afe49b055a476e9f4983e15fcdccac.service
          Unit: user@1000.service
     User Unit: app-org.kde.konsole@d0afe49b055a476e9f4983e15fcdccac.service
         Slice: user-1000.slice
     Owner UID: 1000 (user)
       Boot ID: 73d2e1dd5fe74d5abe316757747c2dcd
    Machine ID: f7e0cfa0e1cc42f1ad1e8a567a167f3c
      Hostname: manjaro
       Storage: /var/lib/systemd/coredump/core.ckb-next\x2ereal.1000.73d2e1dd5fe74d5abe316757747c2dcd.43418.1760963148000000.zst (present)
  Size on Disk: 5.5M
       Message: Process 43418 (ckb-next.real) of user 1000 dumped core.
                
                Stack trace of thread 43418:
                #0  0x00007f6a6ae20ac3 n/a (libffi.so.8 + 0x7ac3)
                #1  0x00007f6a6ae1d76b n/a (libffi.so.8 + 0x476b)
                #2  0x00007f6a6ae2006e ffi_call (libffi.so.8 + 0x706e)
                #3  0x00007f6a6b2cf48d n/a (libwayland-client.so.0 + 0x448d)
                #4  0x00007f6a6b2d02e9 n/a (libwayland-client.so.0 + 0x52e9)
                #5  0x00007f6a6b2d06f3 wl_display_dispatch_queue_pending (libwayland-client.so.0 + 0x56f3)
                #6  0x00007f6a6b239536 _ZN15QtWaylandClient15QWaylandDisplay13flushRequestsEv (libQt6WaylandClient.so.6 + 0x6e536)
                #7  0x00007f6a695c1a74 _ZN7QObject5eventEP6QEvent (libQt6Core.so.6 + 0x1c1a74)
                #8  0x00007f6a6a701dd0 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent (libQt6Widgets.so.6 + 0x101dd0)
                #9  0x00007f6a69568678 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent (libQt6Core.so.6 + 0x168678)
                #10 0x00007f6a69568a5b _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData (libQt6Core.so.6 + 0x168a5b)
                #11 0x00007f6a698437f8 n/a (libQt6Core.so.6 + 0x4437f8)
                #12 0x00007f6a68508880 n/a (libglib-2.0.so.0 + 0x5e880)
                #13 0x00007f6a68509cd7 n/a (libglib-2.0.so.0 + 0x5fcd7)
                #14 0x00007f6a68509ee5 g_main_context_iteration (libglib-2.0.so.0 + 0x5fee5)
[... 371 lines cut ...]
[Current thread is 1 (Thread 0x7f6a64ae69c0 (LWP 43418))]
(gdb) bt
#0  0x00007f6a6ae20ac3 in ?? () from /usr/lib/libffi.so.8
#1  0x00007f6a6ae1d76b in ?? () from /usr/lib/libffi.so.8
#2  0x00007f6a6ae2006e in ffi_call () from /usr/lib/libffi.so.8
#3  0x00007f6a6b2cf48d in ?? () from /usr/lib/libwayland-client.so.0
#4  0x00007f6a6b2d02e9 in ?? () from /usr/lib/libwayland-client.so.0
#5  0x00007f6a6b2d06f3 in wl_display_dispatch_queue_pending () from /usr/lib/libwayland-client.so.0
#6  0x00007f6a6b239536 in QtWaylandClient::QWaylandDisplay::flushRequests() () from /usr/lib/libQt6WaylandClient.so.6
#7  0x00007f6a695c1a74 in QObject::event(QEvent*) () from /usr/lib/libQt6Core.so.6
#8  0x00007f6a6a701dd0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt6Widgets.so.6
#9  0x00007f6a69568678 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt6Core.so.6
#10 0x00007f6a69568a5b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /usr/lib/libQt6Core.so.6
#11 0x00007f6a698437f8 in ?? () from /usr/lib/libQt6Core.so.6
#12 0x00007f6a68508880 in ?? () from /usr/lib/libglib-2.0.so.0
#13 0x00007f6a68509cd7 in ?? () from /usr/lib/libglib-2.0.so.0
#14 0x00007f6a68509ee5 in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#15 0x00007f6a6983ffe2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/libQt6Core.so.6
#16 0x00007f6a69574ca6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt6Core.so.6
#17 0x00007f6a6956cd21 in QCoreApplication::exec() () from /usr/lib/libQt6Core.so.6
#18 0x0000559739c0903c in main ()
(gdb) 
```

Retrograde-i486, 2025-10-22:
I'm getting the same crash as the OP, running it on Garuda Linux, Wayland session, Linux 6.17.4-zen2-1-zen Kernel. Happy to help with crash reports / whatever I can.

tatokis, 2025-10-24:
I think I figured out what triggers it.

Do you have a DisplayPort monitor? Does the GUI crash whenever your monitor goes to sleep? Alternatively, do you press the power button on your monitor to turn it off (standby) and then it crashes?

I can consistently trigger a crash by doing
```
$ WAYLAND_DISPLAY=wayland-1 wlr-randr --output HEADLESS-1 --on
$ WAYLAND_DISPLAY=wayland-1 wlr-randr --output HEADLESS-1 --off
$ 19:21:48.765 [swaybar/tray/host.c:57] Unregistering Status Notifier Item ':1.209/StatusNotifierItem'
19:21:48.765 [swaybar/tray/host.c:57] Unregistering Status Notifier Item ':1.210/StatusNotifierItem'
[7]+  Segmentation fault         (core dumped) WAYLAND_DISPLAY=wayland-1 ckb-next
```

Frankkz, 2025-10-24:
My monitor is connected via DP. The ckb-next GUI is indeed gone after I interact with my system once the monitor went in like a standby mode.